gazebosim/gz-msgs #55: Scene CreateAxisVisual message equivalent?

Levi-Armstrong, 2020-05-17:
I was looking for an equivalent message to use for creating visual axis in my application, but it appears one does not exist. I do see an axis message but I believe it is related to joint axis not a visual axis. I would like to create a message for this visual type and contribute back but wanted to reach out to see what name the maintainers would like to used to represent a scene visual axis?

chapulina, 2020-05-18:
You're talking about [ignition::rendering::Scene::CreateAxisVisual](https://github.com/ignitionrobotics/ign-rendering/blob/a7125d025d3ea6b4786e5b69d286075b800b9d0b/include/ignition/rendering/Scene.hh#L746-L749), right?

> I do see an axis message but I believe it is related to joint axis not a visual axis.

Yes, [axis.proto](https://github.com/ignitionrobotics/ign-msgs/blob/master/proto/ignition/msgs/axis.proto) is for joints.

> I would like to create a message for this visual type and contribute back but wanted to reach out to see what name the maintainers would like to used to represent a scene visual axis?

The intended usage is for the user to send a message and an XYZ axis shows up in the scene, right? Do you have an idea of what parameters it would have? RViz's [axes display](http://wiki.ros.org/rviz/DisplayTypes/Axes) exposes length and radius. I think maybe `visual_axis` / `VisualAxis` could be a good name? And it could probably be added as a marker type:

https://github.com/ignitionrobotics/ign-msgs/blob/e54b5532cb375d65eedab8bd7d9db6124a225966/proto/ignition/msgs/marker.proto#L33-L49

Maybe @iche033 has other ideas.

Levi-Armstrong, 2020-05-18:
> You're talking about ignition::rendering::Scene::CreateAxisVisual, right?

That is correct.

> The intended usage is for the user to send a message and an XYZ axis shows up in the scene, right? 

That is correct. I image it would just include a pose and scale, but I will double check what options  the VisualAxis scene type has.

iche033, 2020-05-18:
It maybe easier to user [makers](https://github.com/ignitionrobotics/ign-gazebo/tree/master/examples/standalone/marker) since that's already been set up in ign-gazebo. If it's just drawing 3 cylinders to represent the axes, markers should be to do the job. But if you need to draw arrows, we'll need to add a `cone` type to markers.

Levi-Armstrong, 2020-05-19:
@iche033 From what I understand the scene method CreateAxisVisual already handles drawing the axis with arrows, should this method be used instead of manually creating cylinders and cones?

iche033, 2020-05-19:
I just noticed this [TODO](https://github.com/ignitionrobotics/ign-rendering/blob/master/ogre2/src/Ogre2Scene.cc#L293) notes in the implementation so it's not working yet :( So you would need to use drawing cylinders and cones for now. 

I'll ticket an issue for the missing implementation

Levi-Armstrong, 2020-05-23:
I do not believe the TODO is valid anymore since it is implemented in the base class for VisualArrow and VisualAxis. As long as the derived class implements cylinder and cone creation then arrow and axis come for free. Do you agree?

iche033, 2020-05-23:
ignitionrobotics/ign-rendering#87 removes the TODO note and added hooks to make Axis/ArrowVisuals work in ogre2

azeey, 2023-10-11:
Closing this in favor of https://github.com/gazebosim/gz-rendering/issues/493

closed 2023-10-11 by azeey