gazebosim/gz-msgs #72: Installation Error:include could not find load file

Anusha-jain-h-c, 2020-07-18:
As following the instructions to run gazebo i came across this :
C:\Users\abdhi\ign-msgs>cd /d C:\Users\abdhi\ign-msgs\

C:\Users\abdhi\ign-msgs>md build

C:\Users\abdhi\ign-msgs>cd build

C:\Users\abdhi\ign-msgs\build>cmake .. -G "NMake Makefiles" -DCMAKE_INSTALL_PREFIX=""install\Release"" -DCMAKE_BUILD_TYPE="Release" -DBUILD_TESTING:BOOL=False
-- The C compiler identification is MSVC 19.26.28806.0
-- The CXX compiler identification is MSVC 19.26.28806.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx86/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx86/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Users/abdhi/ign-cmake/build/ignition-cmake0-config.cmake:47 (include):
  include could not find load file:

    IgnCMake
Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

CMake Error at CMakeLists.txt:17 (ign_configure_project):
  Unknown CMake command "ign_configure_project".

-- Configuring incomplete, errors occurred!
See also "C:/Users/abdhi/ign-msgs/build/CMakeFiles/CMakeOutput.log".

traversaro, 2020-07-18:
It seems that for some reason the CMake project is finding the file `C:/Users/abdhi/ign-cmake/build/ignition-cmake0-config.cmake` that is not meant to be found by CMake and is not working in that location. I do not know why that happened, but if you set the `ignition-cmake0_DIR` in your CMake project to point to the directory that contains `ignition-cmake0-config.cmake` in the prefix where you installed `ignition-cmake0`, everything should work fine.

Anusha-jain-h-c, 2020-07-19:
Thanks.I just reinstalled everything again got some other error...As i am new i cant solve the error.
C:\Users\abdhi\ign-msgs>cd build

C:\Users\abdhi\ign-msgs\build>cmake .. -G "NMake Makefiles" -DCMAKE_INSTALL_PREFIX="" -DCMAKE_BUILD_TYPE="Release" -DBUILD_TESTING:BOOL=False
CMake Error at CMakeLists.txt:12 (find_package):
  By not providing "Findignition-cmake0.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-cmake0", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-cmake0"
  (requested version 0.3.0) with any of the following names:

    ignition-cmake0Config.cmake
    ignition-cmake0-config.cmake

  Add the installation prefix of "ignition-cmake0" to CMAKE_PREFIX_PATH or
  set "ignition-cmake0_DIR" to a directory containing one of the above files.
  If "ignition-cmake0" provides a separate development package or SDK, be
  sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "C:/Users/abdhi/ign-msgs/build/CMakeFiles/CMakeOutput.log".

closed 2020-07-20 by Anusha-jain-h-c

traversaro, 2020-07-20:
> -DCMAKE_INSTALL_PREFIX=""

Hi @Anusha-jain-h-c , I think you need to provide a proper installation directory, as installing directly in the `build` folder is probably not support and bound to the problems you are experiencing.

Anusha-jain-h-c, 2020-07-20:
> > -DCMAKE_INSTALL_PREFIX=""
> 
> Hi @Anusha-jain-h-c , I think you need to provide a proper installation directory, as installing directly in the `build` folder is probably not support and bound to the problems you are experiencing.

Thanks a lot

traversaro, 2020-07-20:
Which `CMAKE_INSTALL_PREFIX` did you passed when installing `ignition-cmake0` ? To which instructions are you referring?

chapulina, 2020-07-20:
Related to https://github.com/ignitionrobotics/ign-fuel-tools/issues/88 and https://github.com/ignitionrobotics/ign-math/issues/138.

I think @Anusha-jain-h-c is following this tutorial: http://gazebosim.org/tutorials?tut=install_on_windows&cat=install

The solution for one of these issues will probably solve them all. It looks like it may be just a matter of updating the documentation.

chapulina, 2020-07-20:
There are some open PRs to fixing Windows tutorials that may be helpful here:

* https://github.com/osrf/gazebo_tutorials/pull/101
* https://github.com/osrf/gazebo_tutorials/pull/97

Arpit-ai, 2021-04-15:
Hi I am getting the same error while i exactly followed the instruction given at: http://gazebosim.org/tutorials?tut=install_on_windows&cat=install

DunZek, 2021-07-18:
Getting the same error still http://gazebosim.org/tutorials?tut=install_on_windows&cat=install

I did not deviate from any of the instructions

cnrgl, 2021-11-22:
i solved issue with added path as PATH option in find_package().  also i tried prefix path but not working for me

CSUN1997, 2021-12-01:
Thanks, @cnrgl for your reply. Your method actually worked for me. Here are the steps that I followed:
1. open CMakeLists.txt under ign-math folder
2. in the function `find_package(ignition-cmake0 0.3.0 REQUIRED)`, add argument `PATHS $your path to ign-cmake release folder$`. In my case, the line is modified to `find_package(ignition-cmake0 0.3.0 REQUIRED PATHS D:\\gz-ws\\ign-cmake\\build\\install\\Release)`.
3. rerun configure.bat in ign-math folder and everything worked

Hope this helps