gazebosim/gz-msgs #152: What is the purpose of ignition::msgs::Double_V?

rahulbhadani, 2021-04-03:
What is the best use case for ignition::msgs::Double_V?

I can see the documentation at https://ignitionrobotics.org/api/msgs/5.3/classignition_1_1msgs_1_1Double__V.html#af5fc2c48339eb12a5c2c09eeecf19ca5

but it doesn't tell me where to use this kind of message?

mjcarroll, 2021-04-05:
The `Double_V` is currently used is to support a "tagged union" variant implementation in the `Packet` message: https://github.com/ignitionrobotics/ign-msgs/blob/1fe993929c815c31c7c38b3b0307782034367d9a/proto/ignition/msgs/packet.proto

This is largely an artifact of how protobuf itself works, which doesn't allow you to have a `repeated` field in a `oneof` structure, and does not allow you to `repeated` a `oneof` structure.

If you are looking to create an array of doubles in your own message, `repeated double` should be sufficient.

mjcarroll, 2021-04-05:
In the future, I believe this type of question is better suited to [Gazebo Answers](https://answers.gazebosim.org/questions/)

closed 2021-04-05 by mjcarroll