gazebosim/gz-msgs #240: PointCloudPackedUtils_TEST.cc fails to compile on Windows

Tobias-Fischer, 2022-04-06:
## Environment
* OS Version: Windows 10
* Source or binary build?
  Build from sources, version 8.3.0; builds available in https://github.com/conda-forge/libignition-msgs1-feedstock/pull/69

## Description
* Expected behavior: Compilation of the test succeeds and tests pass
* Actual behavior: Compilation fails

## Steps to reproduce

1. On Windows, follow https://github.com/regro-cf-autotick-bot/libignition-msgs1-feedstock/blob/8_8.3.0_h8aa9dd/recipe/bld.bat

## Output

```
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(63): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(77): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(82): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(87): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(92): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(97): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/detail/PointCloudPackedUtils.hh(102): error C2977: 'ignition::msgs::PointCloudPackedConstIterator<float>': too many template arguments
%SRC_DIR%\src\PointCloudPackedUtils_TEST.cc(196): note: see declaration of 'ignition::msgs::PointCloudPackedConstIterator<float>'
%SRC_DIR%\include\ignition/msgs/PointCloudPackedUtils.hh(97): error C2512: 'ignition::msgs::PointCloudPackedIteratorBase<FieldType,const FieldType,const char,const ignition::msgs::PointCloudPacked,ignition::msgs::PointCloudPackedConstIterator>': no appropriate default constructor available
        with
        [
            FieldType=float
        ]
%SRC_DIR%\include\ignition/msgs/PointCloudPackedUtils.hh(86): note: see declaration of 'ignition::msgs::PointCloudPackedIteratorBase<FieldType,const FieldType,const char,const ignition::msgs::PointCloudPacked,ignition::msgs::PointCloudPackedConstIterator>'
        with
        [
            FieldType=float
        ]
%SRC_DIR%\include\ignition/msgs/PointCloudPackedUtils.hh(97): error C2436: '{ctor}': member function or nested class in constructor initializer list
NMAKE : fatal error U1077: 'D:\bld\libignition-msgs8_1648178860523\_build_env\Library\bin\cmake.exe' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
```

/cc @traversaro

mjcarroll, 2022-04-11:
@traversaro do you have any plans at looking at this?

traversaro, 2022-04-12:
> @traversaro do you have any plans at looking at this?

I would like, but probably I would not be able to do so in the next days. Anyhow, to make sure that we avoid duplicating efforts I will drop a message here as soon as I start looking into it, so if anyone wants to tackle this before I am able to do so please feel free to do it.

traversaro, 2022-04-15:
I am now looking into this.

traversaro, 2022-04-15:
I do not have any problem locally by compiling with VS2019.

Tobias-Fischer, 2022-04-18:
We resolved this by using VS2019 instead of VS2017 - thus closing. Thanks for looking into this @traversaro!

closed 2022-04-18 by Tobias-Fischer