gazebosim/gz-msgs #297: gz-msgs9 building correctly, but not detecting by system

sachinkum0009, 2022-08-22:
## Environment
* OS Version:  Ubuntu 20.04
* Source or binary build?
  
  
[Source](https://github.com/gazebosim/gz-msgs/tree/gz-msgs9)

## Description
* Expected behavior: 
I am trying to install gz-msgs9
**Used these commands to build**
```bash
mkdir build
cd build
cmake ..
make -j8
sudo make install
```

Build is successfull 

But as I try to compile [gz-transport12](https://github.com/gazebosim/gz-transport.git). Getting following error log
```bash
By not providing "Findgz-msgs9.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gz-msgs9",
  but CMake did not find one.

  Could not find a package configuration file provided by "gz-msgs9" with any
  of the following names:

    gz-msgs9Config.cmake
    gz-msgs9-config.cmake

  Add the installation prefix of "gz-msgs9" to CMAKE_PREFIX_PATH or set
  "gz-msgs9_DIR" to a directory containing one of the above files.  If
  "gz-msgs9" provides a separate development package or SDK, be sure it has
  been installed
```

* Actual behavior: 
After successful build, it should have detected by system.

Please help me. I am not sure if this issue is related to the gz-msgs9 branch itself or something else.

mjcarroll, 2022-08-23:
I believe by default, that msgs will be installed into the `/usr/local` prefix.  You need to make sure that it's in your `CMAKE_PREFIX_PATH` and your `PKG_CONFIG_PATH` in order for it to be picked up by the next package.  You can check these variables by using `env`

For this type of application, I would alternatively recommend using something like `colcon` and `vcs` to build the whole workspace.  We have this outlined in the source install instructions here: https://gazebosim.org/docs/fortress/install_ubuntu_src

sachinkum0009, 2022-08-24:
Hi, thank you for your response. 
You are right, the installation path is `/usr/local`, this is the log when installing 

```bash
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/hydra.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/friction.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/wireless_node.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/pose_animation.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/topic_info.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/duration.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/joint_trajectory.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/parameter_value.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/camera_lens.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/header.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/surface.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/annotated_oriented_3d_box_v.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/magnetometer_sensor.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/video_record.proto
-- Up-to-date: /usr/local/include/ignition/msgs8/ignition/msgs/rest_logout.proto

```

I am using branch gz-msgs9
it should be  /usr/local/include/gz/msgs9/ignition/msgs instead of  /usr/local/include/ignition/msgs8/ignition/msgs, right?

sachinkum0009, 2022-08-24:
Because the other package requires gz-msgs9 as dependency.

mjcarroll, 2022-08-24:
Yes, I would expect it to be installed in `gz/msgs9`.  There will still be some `ignition` headers as shims to handle deprecation warnings as part of the renaming, but the bulk of the installation would be in `msgs9`.

I find the the version number the most confusing part.  Can you verify the branch with `git branch`?  I know that `ignition-msgs8` is still the default branch, so unless otherwise specified, that is what git will clone by default.

sachinkum0009, 2022-08-24:
Yes, I changed the branch to gz-msgs9, here is the log

```bash
>>cd gz-msgs/
>>git branch
* gz-msgs9
  ign-msgs8
```

mjcarroll, 2022-08-24:
Perhaps try removing and recreating the `build` directory to insure that nothing is cached in the the `CMakeCache.txt`?

sachinkum0009, 2022-08-29:
`gz-sim7` branch solved the issue

closed 2022-08-29 by sachinkum0009