gazebosim/gz-msgs #302: ignition/msgs/test_config.h has failing include statement

scpeters, 2022-09-13:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? ABI checker CI build

## Description
* Expected behavior: ABI checker build should compile all headers successfully
* Actual behavior: ABI checker fails to compile a header

The [include/ignition/msgs/test_config.h](https://github.com/gazebosim/gz-msgs/blob/gz-msgs9/include/ignition/msgs/test_config.h) header file was added in https://github.com/gazebosim/gz-msgs/pull/249, but it redirects to a non-existent header file. I'm guessing that this redirection header should not have been added in the first place?

## Steps to reproduce

1. Create a pull request without API/ABI changes targeting `gz-msgs9` (such as https://github.com/gazebosim/gz-msgs/pull/301)
2. Observe failure, and view [workspace log](https://build.osrfoundation.org/job/ignition_msgs-abichecker-any_to_any-ubuntu_auto-amd64/ws/abi_checker/logs/ign-msgs/X/log.txt/*view*/)

## Output

~~~
In file included from /tmp/suIk835nXK/dump1.h:17:
/usr/local/destination_branch/include/gz/msgs9/ignition/msgs/test_config.h:18:10: fatal error: gz/msgs/test_config.h: No such file or directory
   18 | #include <gz/msgs/test_config.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
~~~

methylDragon, 2022-09-13:
In the other libs I'd add a line to https://github.com/gazebosim/gz-msgs/blob/8e10c8d39073a9180492c9a261a3b3d304a9491f/test/CMakeLists.txt#L7 to also configure a gz version of the test_config file.

Let's do that instead?

methylDragon, 2022-09-13:
Correction:
The redirection target was removed as of https://github.com/gazebosim/gz-msgs/commit/7cf04fc6594634b3262a54c164fcb357756821e2 , so the current test_config redirection header leads nowhere.

Furthermore, since the prefix was removed in the above commit, there is no need for a redirection header in the first place.

For both of those reasons, let's remove the test_config redirection header.

scpeters, 2022-09-14:
`ignition/msgs/test_config.h` removed in #303

closed 2022-09-14 by scpeters