gazebosim/gz-msgs #358: :farmer: `gz-msgs10-debbuilder` failing to find ruby library

Crola1702, 2023-06-28:
## Environment
* OS Version: Ubuntu 20.04, 22.04
* Binary build

## Description
* Expected behavior: [gz-msgs10-debbuilder](https://build.osrfoundation.org/job/gz-msgs10-debbuilder/303/) should pass
* Actual behavior: [gz-msgs10-debbuilder](https://build.osrfoundation.org/job/gz-msgs10-debbuilder/303/) is failing

## Steps to reproduce
1. Run a build in [gz-msgs10-debbuilder](https://build.osrfoundation.org/job/gz-msgs10-debbuilder/303/)
2. See it fail

## Output

Reference build: https://build.osrfoundation.org/job/gz-msgs10-debbuilder/303/

Log output:

```
-- Installing: /home/jenkins/workspace/gz-msgs10-debbuilder/build/gz-msgs/debian/tmp/usr/share/protos/gz/msgs/log_control.proto
make[1]: Leaving directory '/home/jenkins/workspace/gz-msgs10-debbuilder/build/gz-msgs/obj-x86_64-linux-gnu'
   dh_install
dh_install: warning: Cannot find (any matches for) "usr/lib/*/ruby/*/msgs[0-99]*/*" (tried in ., debian/tmp)

dh_install: warning: libgz-msgs10-dev missing files: usr/lib/*/ruby/*/msgs[0-99]*/*
dh_install: error: missing files, aborting
make: *** [debian/rules:6: binary] Error 25
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui --source-option=--include-binaries -j5 failed
Build step 'Execute shell' marked build as failure
```

Crola1702, 2023-06-28:
This could be potentially fixed by https://github.com/gazebo-release/gz-msgs10-release/pull/3

Crola1702, 2023-06-30:
I think this problem may have broke the rest of the debbuilders:

Gz-msgs10 is causing some errors in other debbuilders:
* https://build.osrfoundation.org/job/gz-fuel-tools9-debbuilder/228/
* https://build.osrfoundation.org/job/gz-sim8-debbuilder/398/
* https://build.osrfoundation.org/job/gz-gui8-debbuilder/401/
* https://build.osrfoundation.org/job/gz-launch7-debbuilder/388/
* https://build.osrfoundation.org/job/gz-sensors8-debbuilder/408/
* https://build.osrfoundation.org/job/gz-transport13-debbuilder/303/

All share the same output:

<details>
<summary>
Log output:
</summary>

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gz-msgs10/gz-msgs10-config.cmake:226 (include):
  include could not find requested file:

    /usr/lib/x86_64-linux-gnu/cmake/gz-msgs10/gz-msgs-extras.cmake
Call Stack (most recent call first):
  /usr/share/cmake/gz-cmake3/cmake3/GzFindPackage.cmake:243 (find_package)
  CMakeLists.txt:67 (gz_find_package)
...
Source file was:
int main() { return 0; }
Performing C++ SOURCE FILE Test COMPILER_HAS_DEPRECATED_ATTR succeeded with the following output:
Change Dir: /home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_37cbc/fast && gmake[2]: Entering directory '/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
/usr/bin/gmake  -f CMakeFiles/cmTC_37cbc.dir/build.make CMakeFiles/cmTC_37cbc.dir/build
gmake[3]: Entering directory '/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_37cbc.dir/src.cxx.o
/usr/bin/c++ -DCOMPILER_HAS_DEPRECATED_ATTR  -g -O2 -ffile-prefix-map=/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++17 -o CMakeFiles/cmTC_37cbc.dir/src.cxx.o -c /home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.cxx
/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.cxx: In function ‘int main()’:
/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.cxx:2:33: warning: ‘int somefunc()’ is deprecated [-Wdeprecated-declarations]
    2 |     int main() { return somefunc();}
      |                         ~~~~~~~~^~
/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.cxx:1:37: note: declared here
    1 | __attribute__((__deprecated__)) int somefunc() { return 0; }
      |                                     ^~~~~~~~
Linking CXX executable cmTC_37cbc
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_37cbc.dir/link.txt --verbose=1
/usr/bin/c++ -g -O2 -ffile-prefix-map=/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wl,-Bsymbolic-functions -flto=auto -ffat-lto-objects -flto=auto -Wl,-z,relro  CMakeFiles/cmTC_37cbc.dir/src.cxx.o -o cmTC_37cbc 
gmake[3]: Leaving directory '/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
gmake[2]: Leaving directory '/home/jenkins/workspace/gz-fuel-tools9-debbuilder/build/gz-fuel-tools/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'

Source file was:
__attribute__((__deprecated__)) int somefunc() { return 0; }
    int main() { return somefunc();}
dh_auto_configure: error: cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON -DCMAKE_INSTALL_RUNSTATEDIR=/run "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DCMAKE_BUILD_TYPE=RelWithDebInfo .. returned exit code 1
make[1]: *** [debian/rules:8: override_dh_auto_configure] Error 2
```

</details>

Only jammy is failing per debbuilder.

Gz-msgs10 is showing the same problem as [last time](https://github.com/osrf/buildfarmer/issues/438#issuecomment-1611248442):

```
ake[1]: Leaving directory '/home/jenkins/workspace/gz-msgs10-debbuilder/build/gz-msgs/obj-x86_64-linux-gnu'
   dh_install
dh_install: warning: Cannot find (any matches for) "usr/lib/*/ruby/*/msgs[0-99]*/*" (tried in ., debian/tmp)

dh_install: warning: libgz-msgs10-dev missing files: usr/lib/*/ruby/*/msgs[0-99]*/*
```

azeey, 2023-08-29:
This is passing now.

closed 2023-08-29 by azeey