gazebosim/gz-msgs #390: Add way to compile gz-msgs10 via cross-compilation

traversaro, 2023-09-30:
## Desired behavior

I will like (and probably will work on) a way to compile gz-msgs10 via cross-compilation. The way described in https://github.com/gazebosim/gz-msgs/blob/0472ba0bb5fe39d8a14499155c68746109d9acf7/CMakeLists.txt#L27  is now outdated and not working anymore after the changes in https://github.com/gazebosim/gz-msgs/pull/368 .

## Alternatives considered

Never cross-compile gz-msgs10 again (not feasible at the moment for osx-arm64 builds on conda-forge, where there are no osx-arm64 built machines). 

## Implementation suggestion

Not sure, I need to look into that.

## Additional context

See https://github.com/conda-forge/gz-msgs-feedstock/pull/16#issuecomment-1741772622 .

traversaro, 2023-09-30:
I will probably work on this as part of https://github.com/conda-forge/gz-msgs-feedstock/pull/16, but I wanted to have an issue for this to aid any users that could be interested in this.

azeey, 2023-10-11:
Fixed by #392.

closed 2023-10-11 by azeey