gazebosim/gz-msgs #410: Improve documentation

PerFrivik, 2023-11-15:
Improve documentation. Why can't there be message generation like in ROS. Lets say I want to know what nav_msgs/Odometry is, all I need to do is google the message type and I get this beautiful thing: 

https://docs.ros.org/en/noetic/api/nav_msgs/html/msg/Odometry.html

I tried the same for gz::msgs::Actuators, but did not find anything and had to go far into the source code to find anything at all.

mjcarroll, 2023-11-20:
For versions up to `garden`, the documentation is available in `docs.gazebosim.org`, for example: [Odometry](https://gazebosim.org/api/msgs/9/classgz_1_1msgs_1_1Odometry.html)  or [Actuators](https://gazebosim.org/api/msgs/9/classgz_1_1msgs_1_1Actuators.html)

In `harmonic` forward, since the message generation pipeline has changed, we are missing some of the documentation on generated message types, which has been reported here: https://github.com/gazebosim/gz-msgs/issues/399

Additionally, there is on-going work to improve the documentation site to address things like search engine optimization to easily surface this information.

I'm going to close this as a duplicate of #399 and use that to track this for now.  Thank you for reporting.

closed 2023-11-20 by mjcarroll

PerFrivik, 2023-11-20:
Alright, thank you!