gazebosim/gz-msgs #422: Python test fails when installing with DESTDIR

cottsay, 2024-01-26:
## Environment
* OS Version: Any
* Source or binary build?
   Source @ 2b66954

## Description
* Expected behavior: Tests execute successfully
* Actual behavior: `basic_TEST.py` can't find `gz.msgs10` to import

## Steps to reproduce
1. `mkdir -p build && cd build && cmake .. && make`
2. `make install DESTDIR=/tmp/foobar`
3. `ctest`

## Output
```
The following tests FAILED:
	 13 - basic_TEST.py (Failed)
```

---

The test sets `PYTHONPATH` based on where it anticipates the Python module will be installed to, however:
1. In general, a developer can typically invoke `ctest` prior to installing a project and it should use artifacts from the build directories for testing. The Python tests in gz-math behave like this, for example.
2. The existing prediction doesn't work when `GZ_PYTHON_INSTALL_PATH` is changed, perhaps by setting `USE_SYSTEM_PATHS_FOR_PYTHON_INSTALLATION=ON`.
3. When `DESTDIR` is used to direct `make` where to install files instead of `CMAKE_INSTALL_PREFIX`, the final target directory isn't known to `ctest` at all.
4. By overriding `PYTHONPATH` unconditionally, it's not possible to set `PYTHONPATH` manually prior to invoking `ctest`. Without resorting to using `CMAKE_INSTALL_PREFIX` or modifying the cmake, it's impossible to get the test to pass.

Note that `DESTDIR` is the installation root mechanism used in both Debian and RPM builds, but colcon uses `CMAKE_INSTALL_PREFIX` because `DESTDIR` is a Makefile concept and doesn't typically work on Windows.

azeey, 2024-01-26:
I think we have the same situation in gz-transport as well (https://github.com/gazebosim/gz-transport/pull/441#discussion_r1331859531).

scpeters, 2024-02-28:
I think we could copy the approach from gz-math7:

* https://github.com/gazebosim/gz-math/blob/gz-math7/src/python_pybind11/CMakeLists.txt#L194-L198

closed 2026-11-14 by caguero