gazebosim/gz-msgs #460: Custom gz-msgs example raises "File already exists in database" warnings

romulogcerqueira, 2024-09-10:
## Environment
* OS Version: Ubuntu 22.04
* Source build: gz-msgs10 / commit: 0da35a8b1239c3186e398ac4ed2dbe1762f3aaa4

## Description

When compiling the [generating_custom_msgs](https://github.com/gazebosim/gz-msgs/tree/gz-msgs11/examples/generating_custom_msgs) examples and setting the `GZ_DESCRIPTOR_PATH` environment variable to point to the path of the generated `.gz_desc` file, the following warnings are raised when running the Gazebo stack.

```
[libprotobuf ERROR google/protobuf/descriptor_database.cc:120] File already exists in database: gz/msgs/time.proto
[libprotobuf ERROR google/protobuf/descriptor_database.cc:120] File already exists in database: gz/msgs/header.proto
```

## Steps to reproduce / Output
[image]

azeey, 2024-11-11:
@iche033 to test if this is fixed by https://github.com/gazebosim/gz-msgs/pull/472

iche033, 2024-11-11:
after taking a closer look, I think this is unrelated to #472. Tested and saw that warnings are still printed.