gazebosim/gz-msgs #462: gz msg -l  and similar commands failing in binary build onMac OS Monterey (12.7.6) [amd64]

sreekanthr, 2024-09-13:
Unable to run `gz msg` related commands on  gz-msg 11 (ionic).
Getting error while running any msg command except listing the versions

## Environment
* OS Version: mac OS Monterey (12.7.6) (Intel Mac)
* binary build: 11.0.0~pre1

## Description
* Expected behavior:  `gz msg help` and `gz msg -l` should return valid results 
* Actual behavior: Getting an error
---
## Steps to reproduce

1. Install binaries from the instructions here. 
   ```
   brew tap osrf/simulation;
   brew install gz-ionic
   ```

2. Ensure the correct version is installed:

   $  **gz msg --versions**
   ``` 11.0.0~pre1```

3. Try to run the help or list commands

   $ gz msg help
   $  gz msg -l
   $  gz msg --list
---
## Output

   $  **gz msg --versions**

` 11.0.0~pre1`

   $ gz msg help

`
//usr/local/Cellar/gz-msgs11/11.0.0-pre1_2/lib/ruby/gz/cmdmsgs11.rb:33:in execute: uninitialized constant Cmd::LIBRARY_NAME    (NameError)
      	from /usr/local/bin/gz:309:in <main>`

$  gz msg -l

`/usr/local/Cellar/gz-msgs11/11.0.0-pre1_2/lib/ruby/gz/cmdmsgs11.rb:33:in execute: uninitialized constant Cmd::LIBRARY_NAME  (NameError) from /usr/local/bin/gz:309:in <main>
`  

 $  **gz msg --list**

`/usr/local/Cellar/gz-msgs11/11.0.0-pre1_2/lib/ruby/gz/cmdmsgs11.rb:33:in execute: uninitialized constant Cmd::LIBRARY_NAME    (NameError)
      	from /usr/local/bin/gz:309:in <main>`

scpeters, 2024-11-18:
I can reproduce this. The executable installed to `libexec` is working, but the `gz msg` invocation isn't working via the ruby script. There was a test added in https://github.com/gazebosim/gz-msgs/pull/357, but it invokes the executable directly rather than `gz msg` through ruby.

scpeters, 2025-05-29:
fixed in #516

PR #516 (2025-05-29): Fix gz msgs ruby command
# 🦟 Bug fix

Fixed #462.

## Summary

The `gz msg` command has an incorrect path to the `gz-msgs` standalone executable, so it fails to run. This fixes the path.

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#tes [... cut]
commit e3ad7dbb (2025-05-29): Fix gz msgs ruby command (#516)

Fixes #462.
Test gz msg command in CI.
commit 7a5e3cb2 (2025-07-15): Fix gz msgs ruby command (#516)

Fixes #462.
Test gz msg command in CI.

(cherry picked from commit e3ad7dbb2c5d711b7deb74e7a51710c206f5caa1)
closed 2025-05-29 by scpeters