gazebosim/gz-msgs #493: Spell check the tutorials

caguero, 2025-03-06:
As part of our spring cleaning, we want to detect spell issues in our documentation and fix it when possible. Here's a potential plan to tackle this issue:

1. We suggest to start checking out the main branch.
1. Identify the files that contain documentation. These are typically `.md` (Markdown) or `.md.in` (a Markdown file to be generated by `CMake`).
1. Install in your system a tool to check for spelling errors (e.g.: `aspell` in Ubuntu).
1. Run the spell checks in all the documents. Most of the catches will be false positives due to words like `Protobuf`, `CMake` and other words that aren't typically in an English dictionary. Ignore all of these until you catch a real spell problem.
1. Fix it. Usually these programs recommend a proper word or you can later manually modify the file with the correct English term.
1. Create a pull request with your fixes. [This guide](https://gazebosim.org/docs/all/contributing/#contributing-code) can be helpful. Don't be shy, we'll be kind during the process and help you if you don't have much experience with Gazebo yet or the instructions are not clear enough.

anshium, 2025-03-07:
I iterated through all the .md and .md.in files in the directory and found two typos in the file [tutorials/message_generation.md](https://github.com/gazebosim/gz-msgs/blob/gz-msgs11/tutorials/message_generation.md) and fixed them - "restrctions" to "restrictions" and "entrypoint" to "entry point".

Creating a PR.

caguero, 2025-03-24:
Thanks for the contribution, @anshium !

PR #494 (2025-03-07): docs: fix typo in tutorials/message_generation.md
Corrected "restrctions" to "restrictions" and "entrypoint" to "entry point" in the documentation.

# 🦟 Bug fix

Fixes #493 

## Summary

Iterated through all the .md and .md.in files. Found 2 typos in the tutorials/message_generation.md file.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [x] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contr [... cut]
closed 2025-03-24 by caguero