gazebosim/gz-msgs #495: gz_msgs_generate_messages_lib doesn't create a static library

EmmanuelMess, 2025-03-07:
The documentation for `gz_msgs_generate_messages_lib` says it creates a static library:

https://github.com/gazebosim/gz-msgs/blob/135f0402d36edd4d206b206cdfb14aa389026c56/cmake/gz_msgs_generate.cmake#L197

But it creates a shared library:

https://github.com/gazebosim/gz-msgs/blob/135f0402d36edd4d206b206cdfb14aa389026c56/cmake/gz_msgs_generate.cmake#L265

Was this a mistake in the documentation? Is there a way to create it as a static library?

caguero, 2025-03-12:
Thanks for reporting it. It looks like a problem in the documentation as we're using `SHARED` to create a shared library. Do you mind to open a pull request targeting `main` to fix the documentation?

EmmanuelMess, 2025-03-12:
> Do you mind to open a pull request targeting main to fix the documentation?

Wait, so there is no way to create a static library with the messages?